dwinhmi/DWIN_DGUS_HMI #3: How to use hardware serial with arduino boards?

sameer-r, 2023-05-19:
I have already designed a board with hardware serial connected to DWIN HMI. Tried using this library but no success. Pleas help.

Thanks

dwinhmi, 2023-05-22:
Can you share the Interfacing Diagram of Your Board and DWIN HMI Let me check the connections.  also make sure that you have set the appropriate baud rate

bondango, 2023-06-26:
> I have already designed a board with hardware serial connected to DWIN HMI. Tried using this library but no success. Pleas help.
> 
> Thanks

Sameer did you find a solution to use hardware serial, as i also dont want to use software serial as Serial1 has been dediacted to display

S-A-L-M-A-N-A-N-S-A-R-I, 2024-06-20:
I made a fork for it please check