dwinhmi/DWIN_DGUS_HMI #5: Question about the use of 0x00 and 0x82 in the sendBuffer for setting LCD screen brightness

hayschan, 2023-05-22:
I am currently reviewing the code in your repository and came across a specific code snippet related to setting the brightness of an LCD screen. I have a few questions regarding the usage of `0x00` and `0x82` in the `sendBuffer` array. Additionally, I would like to know if these values are documented and, if so, which documentation includes them.

Here is the code in question:

https://github.com/dwinhmi/DWIN_DGUS_HMI/blob/ae75ddad21a0110fb5e8b84a58ce1244c7d82d92/src/DWIN.cpp#L74-L79

My specific questions are:

1. What is the purpose or significance of using `0x00` and `0x82` in the `sendBuffer` array?
2. Are these values (`0x00` and `0x82`) necessary for setting the brightness of the LCD screen? If so, what role do they play?
3. Is there any documentation that explains the use of these values in the context of setting LCD screen brightness?

I appreciate any clarification you can provide on this matter. Thank you for your time and contribution to the repository.

dwinhmi, 2023-05-23:
0x0082 is the memory address of Brightness regsiter 
[image]

You can refer the Development Guide for deep understanding

https://www.dwin-global.com/uploads/T5L_DGUSII-Application-Development-Guide-V2.6.pdf

dwinhmi, 2023-05-25:
Closing the thread as it's not an issue

closed 2023-05-25 by dwinhmi