dwinhmi/DWIN_DGUS_HMI #6: Porting Arduino Library to ESP-IDF Component for DWIN DGUS HMI Display

hayschan, 2023-05-25:
Dear DWIN DGUS HMI Library maintainers,

First of all, I would like to express my appreciation for the excellent work on the [DWIN_DGUS_HMI](https://github.com/dwinhmi/DWIN_DGUS_HMI/) Arduino library. It has been incredibly helpful for integrating DWIN DGUS T5L HMI displays into Arduino projects.

I'm writing to inquire whether there are any plans to port the Arduino library to an ESP-IDF component. As you may know, ESP-IDF is the official development framework for Espressif's ESP32 and ESP8266 microcontrollers. By transforming the library into an ESP-IDF component, it could be easily integrated into ESP-IDF-based projects and even be posted to the ESP Registry (https://components.espressif.com/) for broader accessibility and simplified installation.

The ESP-IDF platform has gained significant popularity among developers due to its robustness and extensive features. Many developers are now migrating their projects from Arduino to ESP-IDF to take advantage of its advanced capabilities. Having the DWIN DGUS HMI library as an ESP-IDF component would greatly benefit the ESP community and facilitate seamless integration of DWIN DGUS T5L HMI displays into ESP-IDF projects.

I understand that porting the library to ESP-IDF may require additional effort and adjustments, but the potential benefits in terms of increased adoption and ease of use would be substantial.

I would be more than willing to contribute to this effort by assisting in the porting process, testing, or providing any necessary support. Moreover, I believe that many other ESP-IDF developers would be excited to contribute as well.

Thank you for your attention to this matter. I look forward to hearing your thoughts on the possibility of porting the DWIN_DGUS_HMI library to an ESP-IDF component and making it available through the ESP Registry.

dwinhmi, 2023-05-25:
Yes, we already planned for it and are currently doing Beta testing for the same. Thanks for your enthusiasm towards working on our open source library till the time you can reach out to us so we can share the early access for the same tejeet@dwin.com.cn

closed 2023-05-25 by dwinhmi

hayschan, 2023-05-25:
Thank you for your prompt response and for considering the porting of the DWIN DGUS HMI Library to an ESP-IDF component. I'm thrilled to hear that you have already planned for it and are currently conducting Beta testing.

I would be delighted to assist in any way possible to expedite the process and ensure a successful integration. I appreciate the opportunity to gain early access to the ESP-IDF version of the library.

In the meantime, I kindly request an estimated time of arrival (ETA) for the completion of the ESP-IDF version of the library. Understanding the expected timeline will help me align my own development plans accordingly and anticipate the availability of the updated library.

If there are any specific areas where you would like additional testing or assistance, please let me know. I would be more than happy to contribute by testing the library on different ESP-IDF platforms, providing feedback, or assisting with documentation and examples.

Once again, thank you for your dedication and openness to collaborating with the community. I'm excited to work together on this project and contribute to the success of the DWIN DGUS HMI Library for ESP-IDF.

hayschan, 2024-01-19:
> Yes, we already planned for it and are currently doing Beta testing for the same.

Any update on the library for ESP-IDF. It's been eight months.