dwinhmi/DWIN_DGUS_HMI #17: Danger: high voltage for ESP32 and others below 5V

rtek1000, 2024-03-09:
The DMG80480Y070_02NN display is releasing 5V at output TX2.

But Datasheet says:
- Output Voltage (TXD): 3.2 Typ.

[image]

rtek1000, 2024-03-10:
I used a resistive divider.
 
```
 TX (DWIN)
  |
  |
 [ ] 1k
  |
  |
  o------ RX (ESP32/ESP8266)
  |
  |
 [ ] 2k2
  |
  |
 GND
```

Xilebo123, 2026-10-06:
Good catch on the TX2 voltage discrepancy. I've been down this rabbit hole too — the DWIN datasheets are inconsistent across variants, and it's easy to grab the wrong one for your specific model suffix.

I ended up building a small catalog (https://dwin-hmi.eu) that indexes all 382 DWIN display models with their exact datasheet PDFs, so you can pull the spec sheet for the precise variant you have (e.g. DMG80480Y070_02NN vs _03WTC) and compare pin voltages side by side. Independent project, not DWIN-affiliated.

Worth double-checking: the "_02NN" suffix on your unit may correspond to a different TXD drive level than the "_03WTC" variants. If you can share your exact model string (the full part number on the back label), it might help narrow down which datasheet actually applies.